PX4/homebrew-px4 #13: Missing explicit gcc-arm-none-eabi-54.rb breaks brew

neonsoftware, 2016-12-28:
Hi 😄 

commit https://github.com/PX4/homebrew-px4/commit/5290c657c3d0c35f9a271a2eed45bb4c07237058 , by removing the gcc-arm-none-eabi-54.rb, causes problems with brew where it was explicitly installed. 
(And I guess also in cascade anyone who installed any package depending on `gcc-arm-none-eabi-54`)

`
Error: No available formula with the name "gcc-arm-none-eabi-54"
`
starts to appear and block most brew commands.
The real problem is that brew cannot even remove the package itself without the formula, so its stuck. 

`
Cellar >>> brew uninstall --force gcc-arm-none-eabi-54
Error: No available formula with the name "gcc-arm-none-eabi-54"
`

I added further in my comment to the commit  https://github.com/PX4/homebrew-px4/commit/5290c657c3d0c35f9a271a2eed45bb4c07237058  (when I still thought I could have strip it out of brew easily and fast)

Thanks in advance :+1

LorenzMeier, 2016-12-28:
Fixed in commit 71c3f78 - let me know if that doesn't fix it for you after updating.

closed 2016-12-28 by LorenzMeier

neonsoftware, 2016-12-28:
Almost there 😄 

`
Error: No available formula with the name "gcc-arm-none-eabi-54"
In formula file: /usr/local/Homebrew/Library/Taps/px4/homebrew-px4/gcc-arm-none-eabi-54.rb
Expected to find class GccArmNoneEabi54, but only found: GccArmNoneEabi.
`

LorenzMeier, 2016-12-28:
Sorry, stupid mistake - fixed in a300025

neonsoftware, 2016-12-28:
Works.  You win at fast-fix, sir. 👍