PX4/homebrew-px4 #15: Warning: Calling BottleSpecification.revision is deprecated!

plusk01, 2017-08-06:
```
$ brew tap PX4/homebrew-px4
...
==> Tapping px4/px4
Cloning into '/usr/local/Homebrew/Library/Taps/px4/homebrew-px4'...
remote: Counting objects: 16, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 16 (delta 5), reused 6 (delta 3), pack-reused 0
Unpacking objects: 100% (16/16), done.
Warning: Calling BottleSpecification.revision is deprecated!
Use BottleSpecification.rebuild instead.
/usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb:21:in `block in <class:Qt54>'
Please report this to the px4/px4 tap!

Tapped 13 formulae (52 files, 47.4KB)

$ brew --version
Homebrew 1.3.0
Homebrew/homebrew-core (git revision 24d8; last commit 2017-08-06)
```

Seems that the `revision 1` [line](https://github.com/PX4/homebrew-px4/blob/master/qt54.rb#L21) in `qt54.rb:21` needs to be changed to `rebuild 1`.

(Relevant issue from homebrew-php: https://github.com/Homebrew/homebrew-php/issues/4344)

dgentry, 2017-10-19:
Same thing with a newer version of homebrew:

```
==> Tapping px4/px4
Cloning into '/usr/local/Homebrew/Library/Taps/px4/homebrew-px4'...
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 17 (delta 6), reused 7 (delta 4), pack-reused 0
Unpacking objects: 100% (17/17), done.
Warning: Calling BottleSpecification.revision is deprecated!
Use BottleSpecification.rebuild instead.
/usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb:21:in `block in <class:Qt54>'
Please report this to the px4/px4 tap!

Tapped 14 formulae (54 files, 48.8KB)
Already up-to-date.

$ brew --version
Homebrew 1.3.6
Homebrew/homebrew-core (git revision e7ed31; last commit 2017-10-18)
```

BrandonOsas, 2018-01-19:
now its just straight error, not even a warning
==> Tapping px4/px4
Cloning into '/usr/local/Homebrew/Library/Taps/px4/homebrew-px4'...
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 17 (delta 5), reused 5 (delta 2), pack-reused 0
Unpacking objects: 100% (17/17), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb
Calling BottleSpecification.revision is disabled!
Use BottleSpecification.rebuild instead.
/usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb:21:in `block in <class:Qt54>'
Please report this to the px4/px4 tap!
Or, even better, submit a PR to fix it!
Error: Cannot tap px4/px4: invalid syntax in tap!

BrandonOsas, 2018-01-19:
now its just straight error, not even a warning

> ==> Tapping px4/px4
Cloning into '/usr/local/Homebrew/Library/Taps/px4/homebrew-px4'...
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 17 (delta 5), reused 5 (delta 2), pack-reused 0
Unpacking objects: 100% (17/17), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb
Calling BottleSpecification.revision is disabled!
Use BottleSpecification.rebuild instead.
/usr/local/Homebrew/Library/Taps/px4/homebrew-px4/qt54.rb:21:in `block in <class:Qt54>'
Please report this to the px4/px4 tap!
Or, even better, submit a PR to fix it!
Error: Cannot tap px4/px4: invalid syntax in tap!

LorenzMeier, 2018-02-03:
We did delete the bottle on our side as it was not being used any more, closing.

closed 2018-02-03 by LorenzMeier