PX4/homebrew-px4 #16: Unable to install fastrtps

meteor0530, 2017-09-01:
```
==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
######################################################################## 100.0%
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/Cellar/fastrtps/1.5.0 ..
==> make
Last 15 lines from /Users/meteor/Library/Logs/Homebrew/fastrtps/02.make:
[ 86%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/builtin/data/ReaderProxyData.cpp.o
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputController.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputControllerDescriptor.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/FlowController.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/exceptions/Exception.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/attributes/PropertyPolicy.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/common/Token.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParserCommon.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileElementParser.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileParser.cpp.o
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

```

fabianschilling, 2017-09-28:
==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps, fastcdr
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
Already downloaded: /Users/fabian/Library/Caches/Homebrew/fastrtps-1.5.0.tar.gz
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/Cellar/fastrtps/1.5
Last 15 lines from /Users/fabian/Library/Logs/Homebrew/fastrtps/01.cmake:

CMake Error at /usr/local/Cellar/cmake/3.9.3_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Java: Found unsuitable version "..", but required is at
  least "1.6" (found
  /Library/Java/JavaVirtualMachines/jdk-9.jdk/Contents/Home/bin/java)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.9.3_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:375 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.9.3_1/share/cmake/Modules/FindJava.cmake:243 (find_package_handle_standard_args)
  cmake/dev/java_support.cmake:16 (find_package)
  CMakeLists.txt:226 (gradle_build)

KochC, 2017-10-05:
Had the same issues. Solved it by downgrading to **Java 1.6** to solve the issue described by @fabianschilling 

> Could NOT find Java: Found unsuitable version "..", but required is at
  least "1.6"

Also solved the issue from @meteor0530 by installing Java 1.8 from [here](http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html) which is the source described in the manual.

frank-rui, 2017-10-05:
brew install px4-dev
Updating Homebrew...
==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps, fastcdr
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
Already downloaded: /Users/appleapple/Library/Caches/Homebrew/fastrtps-1.5.0.tar.gz
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local
==> make
Last 15 lines from /Users/appleapple/Library/Logs/Homebrew/fastrtps/02.make:
[ 86%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/builtin/data/ReaderProxyData.cpp.o
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputController.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputControllerDescriptor.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/FlowController.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/exceptions/Exception.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/attributes/PropertyPolicy.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/common/Token.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParserCommon.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileElementParser.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileParser.cpp.o
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/px4/homebrew-px4/issues

These open issues may also help:
Unable to install fastrtps https://github.com/PX4/homebrew-px4/issues/16

sleddriver, 2017-10-09:
I have this issue as well on High Sierra, I have tried to reinstall java and this several times with the same result.

tops4u, 2017-10-26:
Same problem here, also with High Sierra.

tops4u, 2017-11-13:
I think I found a workaround for this Problem. 

_brew cask install java_ 

will install JAVA9 and there is a Class Name Collision (java.lang.Module) so that part of fastrtps build fails and in sequence the linker. I got it working by uninstalling Java 9 and manually installing the JAVA 8 JDK Version from the Oracle site. 

Also there was a Problem installing jinja2 so I had to modify the installation command to : _sudo -H pip install pyserial jinja2_

which worked well for me.

LorenzMeier, 2017-11-13:
Could you please update from Home-brew and see if that resolves your issue? I avoided building with JAVA by adding a new bottle: https://github.com/PX4/homebrew-px4/commit/bd6519ba3f4ce939d2835efc61259489f74cd47a

ChristophTobler, 2017-11-14:
Now works for high-sierra with latest master. @LorenzMeier thanks!

ChristophTobler, 2017-11-14:
@meteor0530 Please reopen if it's still an issue for you.

closed 2017-11-14 by ChristophTobler