PX4/homebrew-px4 #17: Unknown Error During Install - 'make: *** [all] Error 2'

frank-rui, 2017-10-06:
System: MacOS High Sierra 10.13

Java: Installed the lasted Java 9, but I switched to both version 1.7, 1.8 as recommended by @KochC, using command, i.e., switch to java version 1.8 by,

frankmbp:~ appleapple$ export JAVA_HOME=/usr/libexec/java_home -v 1.8
frankmbp:~ appleapple$ java -version
java version "1.8.0_144"
Java(TM) SE Runtime Environment (build 1.8.0_144-b01)
Java HotSpot(TM) 64-Bit Server VM (build 25.144-b01, mixed mode)

-------- The error messages during installation of brew install px4-dev -------------
brew install px4-dev
Updating Homebrew...
==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps, fastcdr
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
Already downloaded: /Users/appleapple/Library/Caches/Homebrew/fastrtps-1.5.0.tar.gz
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local
==> make
Last 15 lines from /Users/appleapple/Library/Logs/Homebrew/fastrtps/02.make:
[ 86%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/builtin/data/ReaderProxyData.cpp.o
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputController.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputControllerDescriptor.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/FlowController.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/exceptions/Exception.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/attributes/PropertyPolicy.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/common/Token.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParserCommon.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileElementParser.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileParser.cpp.o
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir///thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/px4/homebrew-px4/issues

These open issues may also help:
Unable to install fastrtps #16

KochC, 2017-10-08:
Try to switch to **Java 1.6** to compile **fastrtps**. Then switch back later to **Java 1.8**

LorenzMeier, 2017-10-08:
This is annoying - let me see if we can fix this.

frank-rui, 2017-10-08:
Thank you @KochC for the replay. I've tried Java 1.6 but the error still occured.
Below is the error message,
---------------------
frankmbp:~ appleapple$ java -version

java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)

-----

frankmbp:~ appleapple$ brew install px4-dev
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (caskroom/cask, homebrew/core).
==> New Formulae
bzt                        mecab-unidic-extended      precomp
crc32c                     msgpuck
duc                        pdfpc
==> Updated Formulae
augeas              feh                 osm2pgsql           sshguard
aws-sdk-cpp         gwt                 osquery             sysdig
bear                hopenpgp-tools      osrm-backend        vapoursynth
bento4              httpd               pcap_dnsproxy       vnu
blahtexml           ibex                pktanon             x265
cfr-decompiler      imagemagick         pldebugger          xalan-c
checkstyle          imagemagick@6       postgis             xerces-c
cppcheck            jsonnet             prometheus          xml-security-c
dbxml               kitchen-sync        protobuf-swift      xml-tooling-c
dcm2niix            libxml2             pyenv-virtualenv    xqilla
emscripten          links               pypy                xsd
enigma              mpw                 pypy3               xtensor
etcd                mypy                remake              zile
fd                  opensaml            serd                zorba
==> Renamed Formulae
thrift@0.90 -> thrift@0.9

==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps, fastcdr
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
Already downloaded: /Users/appleapple/Library/Caches/Homebrew/fastrtps-1.5.0.tar.gz
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local
==> make
Last 15 lines from /Users/appleapple/Library/Logs/Homebrew/fastrtps/02.make:
[ 86%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/builtin/data/ReaderProxyData.cpp.o
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputController.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputControllerDescriptor.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/FlowController.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/exceptions/Exception.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/attributes/PropertyPolicy.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/common/Token.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParserCommon.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileElementParser.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileParser.cpp.o
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

LorenzMeier, 2017-11-13:
Could you please re-test if after a home-brew update everything is back to normal?

mje-nz, 2017-11-13:
I'm on El Capitan, so I had to use "brew install --force-bottle fastrps" to use your bottle. I can install px4-dev and make px4fmu-v2_default now. Could you please add the bottle for El Capitan?

LorenzMeier, 2017-11-14:
Did that just now. I'm closing this for now, please re-open if it doesn't work.

closed 2017-11-14 by LorenzMeier