PX4/homebrew-px4 #18: Upgrade to latest version fails.

jtilbian, 2017-12-07:
pacifist@MacBook-Pro ~> brew outdated 
px4/px4/gcc-arm-none-eabi-63 (20170223) < 20170628
pacifist@MacBook-Pro ~> brew upgrade px4/px4/gcc-arm-none-eabi-63 
==> Upgrading 1 outdated package, with result:
px4/px4/gcc-arm-none-eabi-63 20170628
==> Upgrading px4/px4/gcc-arm-none-eabi-63 
==> Downloading https://developer.arm.com/-/media/Files/downloads/gnu-rm/6-2017q
==> Downloading from https://armkeil.blob.core.windows.net/developer/Files/downl
######################################################################## 100.0%
==> Copying binaries...
==> cp -r arm-none-eabi bin lib share /usr/local/Cellar/gcc-arm-none-eabi-63/201
Last 15 lines from /Users/pacifist/Library/Logs/Homebrew/gcc-arm-none-eabi-63/01.cp:
2017-12-06 22:39:03 -0800

cp
-r
arm-none-eabi
bin
lib
share
/usr/local/Cellar/gcc-arm-none-eabi-63/20170628/

cp: arm-none-eabi: No such file or directory
cp: bin: No such file or directory
cp: lib: No such file or directory
cp: share: No such file or directory

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/px4/homebrew-px4/issues

pdemil, 2018-01-07:
Same issue here.

LorenzMeier, 2018-02-03:
That's fixed now.

closed 2018-02-03 by LorenzMeier