PX4/homebrew-px4 #26: Cannot install kconfig-frontends

mcsmurf, 2018-07-10:
Does anyone have an alternative site to get this from? I seem to be getting a timeout:
`==> Installing kconfig-frontends from px4/px4
==> Downloading http://ymorin.is-a-geek.org/download/kconfig-frontends/kcon

curl: (7) Failed to connect to ymorin.is-a-geek.org port 80: Operation timed out
Error: Failed to download resource "kconfig-frontends"
Download failed: http://ymorin.is-a-geek.org/download/kconfig-frontends/kconfig-frontends-3.7.0.0.tar.xz`

heliDoc2018, 2018-10-02:
@mcsmurf do you found an solution 4 the config-frontends.3.7.0.0.tar.xz problem ?

roboflot, 2018-10-04:
I faced the same problem today while installing developer's toolchain on my MacOS. 
Any help with alternative is appreciate!

mcsmurf, 2018-10-04:
@heliDoc2018, @roboflot I haven't found a solution yet. I've switched to Ubuntu for the time being.

LorenzMeier, 2018-10-04:
Hi all - sorry for missing this earlier. It should be fixed for all Mac OS versions now. Please ping me again here if not.

LorenzMeier, 2018-10-04:
To fix: ```brew update``` and then the install command as detailed in our instructions.

closed 2018-10-04 by LorenzMeier

mcsmurf, 2018-10-05:
Thanks @LorenzMeier, installation was successful.

akshata-1, 2021-11-23:
Were you able to solve this issue? 
I'm using Ubuntu 18.04 LTS
Error-

E: Unable to locate package kconfig-frontends

Jaeyoung-Lim, 2021-11-23:
@akshata-1 See https://github.com/PX4/PX4-Autopilot/pull/18700 This repo is for running PX4 on macOS and not Ubuntu. Therefore your problem is not related to this PR