PX4/homebrew-px4 #30: issue on brew install px4-dev

skateli, 2018-11-14:
when I launch the command brew install px4-dev on my mac (version :10.14.1), the error showed as below
Firmware skate$ brew install px4-dev
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 3 taps (osrf/simulation, homebrew/core and homebrew/cask).
==> New Formulae
osrf/simulation/ignition-gui1
==> Updated Formulae
checkbashisms                            osrf/simulation/ignition-sensors0
libuv                                    osrf/simulation/sdformat8

==> Installing px4-dev from px4/px4
==> Installing dependencies for px4/px4/px4-dev: fastrtps and fastcdr
==> Installing px4/px4/px4-dev dependency: fastrtps
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-1.5.0.mojave.bottle.t

curl: (22) The requested URL returned error: 403 Forbidden
Error: Failed to download resource "fastrtps"
Download failed: http://px4-tools.s3.amazonaws.com/fastrtps-1.5.0.mojave.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.5.0p.tar.gz
Already downloaded: /Users/skate/Library/Caches/Homebrew/downloads/c137e38e8520bbfa08069b90bbe92248e54b923a847a2a3b3290230451b82201--fastrtps-v1.5.0p.tar.gz
==> Patching
patching file CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 373 with fuzz 1.
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local
==> make
Last 15 lines from /Users/skate/Library/Logs/Homebrew/fastrtps/02.make:
[ 86%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/builtin/data/ReaderProxyData.cpp.o
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputController.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/ThroughputControllerDescriptor.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/flowcontrol/FlowController.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/exceptions/Exception.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/attributes/PropertyPolicy.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/common/Token.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParserCommon.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileElementParser.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileParser.cpp.o
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/px4/homebrew-px4/issues

dagar, 2018-11-14:
@skateli I've added a Mojave bottle. Could you give this another try?

pietrodn, 2018-12-29:
Duplicate of #28, solved by #29. This can be closed.

closed 2018-12-29 by TSC21