PX4/homebrew-px4 #39: Error building stable for px4fmu-v3_default on macOS Mojave

dlwalter, 2019-03-22:
I get the following error when trying to build from stable branch for the px4fmu-v3_default:

```
➜  Firmware git:(stable) make px4fmu-v3_default
CMake Warning (dev) at cmake/configs/nuttx_px4fmu-v3_default.cmake:3 (set):
  implicitly converting 'string' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:151 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/configs/nuttx_px4fmu-v3_default.cmake:4 (set):
  implicitly converting 'string' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:151 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/configs/nuttx_px4fmu-v3_default.cmake:5 (set):
  implicitly converting 'string' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:151 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/configs/nuttx_px4fmu-v3_default.cmake:6 (set):
  implicitly converting 'string' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:151 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- PX4 VERSION: v1.8.2
-- CONFIG: nuttx_px4fmu-v3_default
-- Build Type: MinSizeRel
CMake Deprecation Warning at /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeForceCompiler.cmake:72 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:176 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeForceCompiler.cmake:86 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:176 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeForceCompiler.cmake:72 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  build/nuttx_px4fmu-v3_default/CMakeFiles/3.14.0/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:176 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.14.0/share/cmake/Modules/CMakeForceCompiler.cmake:86 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  build/nuttx_px4fmu-v3_default/CMakeFiles/3.14.0/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:176 (project)

-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/local/anaconda2/bin/python (found version "2.7.14")
-- Found PY_jinja2: /usr/local/anaconda2/lib/python2.7/site-packages/jinja2
-- C compiler: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Submodule 'msg/tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'tools/gencpp'
[... 395 lines cut ...]
                 from ../../src/lib/ecl/attitude_fw/ecl_roll_controller.cpp:41:
NuttX/nuttx/include/arch/math.h:87:20: error: expected initializer before '_PARAMS'
 extern double atan _PARAMS((double));
                    ^~~~~~~
compilation terminated due to -Wfatal-errors.
[21/998] Building CXX object src/lib/ecl/airdata/CMakeFiles/ecl_airdata.dir/WindEstimator.cpp.obj
FAILED: src/lib/ecl/airdata/CMakeFiles/ecl_airdata.dir/WindEstimator.cpp.obj
/usr/local/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_PX4FMU_V2 -DMODULE_NAME=\"ecl/airdata\" -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/lib/ecl -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG -MD -MT src/lib/ecl/airdata/CMakeFiles/ecl_airdata.dir/WindEstimator.cpp.obj -MF src/lib/ecl/airdata/CMakeFiles/ecl_airdata.dir/WindEstimator.cpp.obj.d -o src/lib/ecl/airdata/CMakeFiles/ecl_airdata.dir/WindEstimator.cpp.obj -c ../../src/lib/ecl/airdata/WindEstimator.cpp
In file included from NuttX/nuttx/include/math.h:52:0,
                 from NuttX/nuttx/include/cxx/cmath:46,
                 from ../../src/platforms/px4_defines.h:67,
                 from ../../src/platforms/px4_log.h:136,
                 from ../../src/lib/ecl/ecl.h:44,
                 from ../../src/lib/ecl/airdata/WindEstimator.hpp:41,
                 from ../../src/lib/ecl/airdata/WindEstimator.cpp:39:
NuttX/nuttx/include/arch/math.h:87:20: error: expected initializer before '_PARAMS'
 extern double atan _PARAMS((double));
                    ^~~~~~~
compilation terminated due to -Wfatal-errors.
[27/998] Generating uORB topic sources
FAILED: ../nuttx_px4io-v2_default/nuttx_px4io-v2_default.elf
cd /Users/dlw/Workspace/PX4/Firmware && make --no-print-directory nuttx_px4io-v2_default
ninja: build stopped: subcommand failed.
make: *** [px4fmu-v3_default] Error 1
```

dagar, 2019-03-23:
You either need to change to the exact toolchain used at the time or apply this change. https://github.com/PX4/NuttX/pull/42

From the output you're using GCC 7 2018 q2 update. Going back to v1.8.2 the toolchain was GCC 7 2017 q4 major.

dlwalter, 2019-03-25:
Thanks Daniel.  

Does anyone know a way to switch back to 2017 q4 through homebrew?

dlwalter, 2019-03-25:
Got it.  Thanks.

closed 2019-10-09 by dlwalter