PX4/homebrew-px4 #72: fastrtps error while 'brew install px4-dev'.

darkasassin44, 2021-01-11:
My laptop is MacBook Pro 16"(2020)
and using macOS Catalina 10.15.7 now.

I've been following this guide : https://dev.px4.io/v1.11/en/setup/dev_env_mac.html#common-tools
While doing `brew install px4-dev` , the error below has occurred.

```
 ✘ songunsuk@sams-MacBookPro  ~  brew install px4-dev
==> Installing px4-dev from px4/px4
==> Downloading https://homebrew.bintray.com/bottles/openjdk-15.0.1.catalina.bot
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/355a80454da4f90367777654ba877fba8c22ddfca371cedc39c3ef0724a2b334--openjdk-15.0.1.catalina.bottle.tar.gz
==> Downloading https://www.apache.org/dyn/closer.lua?path=ant/ivy/2.5.0/apache-
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/4db50de19f1cbf8e566630b0b94213336416c6f10533a1e84d094e2f1e2d054a--apache-ivy-2.5.0-bin.tar.gz
==> Downloading https://www.apache.org/dyn/closer.lua?path=commons/bcel/binaries
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/222d3991dff4242f666efb012609c563ee961b6c39dbe96e26dcf156b030c8ba--bcel-6.5.0-bin.tar.gz
==> Downloading https://www.apache.org/dyn/closer.lua?path=ant/binaries/apache-a
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/b981c4f0bf945c4618c4b3ea9d9a163ee6adb924eff92784fa28b2e8f144657e--apache-ant-1.10.9-bin.tar.xz
==> Downloading https://homebrew.bintray.com/bottles/astyle-3.1.catalina.bottle.
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/5d1fce73203cc1c1088892784aebe18a7347e5db872f4ad3bff79655349f8341--astyle-3.1.catalina.bottle.1.tar.gz
==> Downloading https://homebrew.bintray.com/bottles/bash-completion-1.3_3.catal
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/21be184d6ca40d06076f8150eb737bf9e3213ffaeecce6c91c096f06014e7218--bash-completion-1.3_3.catalina.bottle.tar.gz
==> Downloading https://homebrew.bintray.com/bottles/zstd-1.4.8.catalina.bottle.
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/a16872669b3c44c7c203136bae527db54657b773317bc025dfd638294dd7a4e1--zstd-1.4.8.catalina.bottle.tar.gz
==> Downloading https://homebrew.bintray.com/bottles/ccache-4.1.catalina.bottle.
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/9694f882d662d03f258e3f79eb6624a98993659d5e1e01a3fa8a399153b56b5b--ccache-4.1.catalina.bottle.tar.gz
==> Downloading https://github.com/discoteq/flock/releases/download/v0.4.0/flock
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/0199e8c557e3b237f2970315d992e0a8d3ecc765e5651c9aaf30e09a03463e13--flock-0.4.0.tar.xz
==> Downloading https://github.com/eProsima/Fast-CDR/archive/v1.0.8.tar.gz
Already downloaded: /Users/songunsuk/Library/Caches/Homebrew/downloads/c18852789305dc4631a82be63883d74c369848f4de7218481039e80f5d23f3fd--Fast-CDR-1.0.8.tar.gz
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-1.6.0.high_sierra.bot
-=O=-                              #     #     #     #                        
curl: (52) Empty reply from server
Error: Failed to download resource "fastrtps"
Download failed: http://px4-tools.s3.amazonaws.com/fastrtps-1.6.0.high_sierra.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading http://px4-tools.s3.amazonaws.com/fastrtps-v1.6.0.tar.gz
-=O=-                                #     #     #     #                      
curl: (52) Empty reply from server
Error: Failed to download resource "fastrtps"
Download failed: http://px4-tools.s3.amazonaws.com/fastrtps-v1.6.0.tar.gz
 ✘ songunsuk@sams-MacBookPro  ~ 
``` 

To fix this error and proceed, what should I do?

LorenzMeier, 2021-01-16:
I can download this fine - this is most likely related to your own network setup.

closed 2021-01-16 by LorenzMeier

darkasassin44, 2021-01-17:
> I can download this fine - this is most likely related to your own network setup.

Oh, sorry, I forgot to consider the network...
I was in university's own network and there has some network issues about server permission on Linux too.
Now I can get to the server and it is completely fine.
Thank you