PX4/homebrew-px4 #77: macOS 11.5.1 brew install px4-dev fails for fastrtps

NathanDotTo, 2021-08-01:
Please see full error trace below.

```
brew install px4-dev
...
==> Installing px4-dev dependency: fastrtps
/usr/local/Homebrew/Library/Taps/px4/homebrew-px4/Formula/fastrtps.rb:19: warning: conflicting chdir during another chdir block
==> cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/homebrew/Cellar/fastrtps/1.6.0 ..
==> make
Last 15 lines from /Users/nsowatsk/Library/Logs/Homebrew/fastrtps/02.make:
[ 88%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLEndpointParser.cpp.o
[ 89%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLParser.cpp.o
[ 90%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/xmlparser/XMLProfileManager.cpp.o
[ 91%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/PersistentWriter.cpp.o
[ 92%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/StatelessPersistentWriter.cpp.o
[ 93%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/StatefulPersistentWriter.cpp.o
[ 94%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/reader/StatelessPersistentReader.cpp.o
[ 95%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/reader/StatefulPersistentReader.cpp.o
[ 96%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/persistence/PersistenceFactory.cpp.o
[ 97%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/persistence/SQLite3PersistenceService.cpp.o
[ 98%] Building C object src/cpp/CMakeFiles/fastrtps.dir/rtps/persistence/sqlite3.c.o
[ 99%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/tinyxml2/tinyxml2.cpp.o
[100%] Linking CXX shared library libfastrtps.dylib
[100%] Built target fastrtps
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

Traceback (most recent call last):
	26: from /usr/local/homebrew/Library/Homebrew/build.rb:229:in `<main>'
	25: from /usr/local/homebrew/Library/Homebrew/build.rb:133:in `install'
	24: from /usr/local/homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	23: from /usr/local/homebrew/Library/Homebrew/build.rb:138:in `block in install'
	22: from /usr/local/homebrew/Library/Homebrew/formula.rb:1274:in `brew'
	21: from /usr/local/homebrew/Library/Homebrew/formula.rb:2394:in `stage'
	20: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	19: from /usr/local/homebrew/Library/Homebrew/resource.rb:91:in `stage'
	18: from /usr/local/homebrew/Library/Homebrew/resource.rb:116:in `unpack'
	17: from /usr/local/homebrew/Library/Homebrew/resource.rb:199:in `mktemp'
	16: from /usr/local/homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	15: from /usr/local/homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	14: from /usr/local/homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	13: from /usr/local/homebrew/Library/Homebrew/resource.rb:117:in `block in unpack'
	12: from /usr/local/homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	11: from /usr/local/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
	10: from /usr/local/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
	 9: from /usr/local/homebrew/Library/Homebrew/resource.rb:121:in `block (2 levels) in unpack'
	 8: from /usr/local/homebrew/Library/Homebrew/formula.rb:2414:in `block in stage'
	 7: from /usr/local/homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	 6: from /usr/local/homebrew/Library/Homebrew/formula.rb:2415:in `block (2 levels) in stage'
	 5: from /usr/local/homebrew/Library/Homebrew/formula.rb:1281:in `block in brew'
	 4: from /usr/local/homebrew/Library/Homebrew/build.rb:178:in `block (2 levels) in install'
	 3: from /usr/local/Homebrew/Library/Taps/px4/homebrew-px4/Formula/fastrtps.rb:21:in `install'
	 2: from /usr/local/homebrew/Library/Homebrew/formula.rb:2185:in `system'
	 1: from /usr/local/homebrew/Library/Homebrew/formula.rb:2185:in `open'
/usr/local/homebrew/Library/Homebrew/formula.rb:2249:in `block in system': Failed executing: make (BuildError)
	5: from /usr/local/homebrew/Library/Homebrew/brew.rb:164:in `<main>'
	4: from /usr/local/homebrew/Library/Homebrew/brew.rb:176:in `rescue in <main>'
	3: from /usr/local/homebrew/Library/Homebrew/exceptions.rb:509:in `dump'
	2: from /usr/local/homebrew/Library/Homebrew/exceptions.rb:455:in `issues'
	1: from /usr/local/homebrew/Library/Homebrew/exceptions.rb:459:in `fetch_issues'
/usr/local/homebrew/Library/Homebrew/utils/github.rb:60:in `issues_for_formula': undefined method `full_name' for nil:NilClass (NoMethodError)

```

AffinityGH, 2021-08-02:
Running into the same issue with an M1 Macbook Air.

Jimwaga, 2021-09-25:
Having the same issues. I tried
1. Duplicating Terminal and in Get info, added Open with Rosetta 2 -> didn't get Traceback errors
2. Installed separate brew, and created alias by moving it to '/usr/local/homebrew' -> generated traceback

Using M1 Macbook Air

christopherramosaz, 2021-09-27:
Same issue. M1 Macbook Pro with a fresh Big Sur 11.6 install.

dlech, 2021-11-10:
It looks like #79 has been proposed to fix this.

ThomasDebrunner, 2021-11-10:
Yes, this issue is caused by running on Apple Silicon (M1). The recommended and supported way is to use rosetta emulation as described in https://docs.px4.io/v1.12/en/dev_setup/dev_env_mac.html.

PR https://github.com/PX4/homebrew-px4/pull/80 should allow installing the all natively on M1 though

edbyy, 2022-06-09:
I still had this problem on 12.4 even after following the recommended steps of using Rosetta emulation. What finally worked for me was using iTerm and running all of the commands (including installing homebrew again) using the `arch -x86_64` prefix.

troy-at-skyfront, 2023-01-17:
> I still had this problem on 12.4 even after following the recommended steps of using Rosetta emulation. What finally worked for me was using iTerm and running all of the commands (including installing homebrew again) using the `arch -x86_64` prefix.

Confirmed this worked for me as well.