PX4/homebrew-px4 #78: brew install px4-sim-gazebo fails on macOS 11.5.1

NathanDotTo, 2021-08-01:
I am not sure where to register this issue I am afraid, so noting it here as the closest approximation.

```
 ~ % brew install px4-sim-gazebo
...
==> Downloading https://ghcr.io/v2/homebrew/core/ode/blobs/sha256:333320201f493ecb42eb9754a8c73d8490aa8d0155129865384fe2faf2706482
Already downloaded: /Users/nsowatsk/Library/Caches/Homebrew/downloads/ec7045e34722cb960dc9207ec13fa50e60a618d00353233d4846bcf93e134f55--ode--0.16.2.big_sur.bottle.tar.gz
Warning: Building gtk+ from source as the bottle needs:
- HOMEBREW_CELLAR: /usr/local/Cellar (yours is /usr/local/homebrew/Cellar)
- HOMEBREW_PREFIX: /usr/local (yours is /usr/local/homebrew)
==> Downloading https://bug757187.bugzilla-attachments.gnome.org/attachment.cgi?id=331173
==> Downloading from https://bugzilla.gnome.org/attachment.cgi?id=331173
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: ce5adf1a019ac7ed2a999efb65cfadeae50f5de8663638c7f765f8764aa7d931
  Actual: b542466d48ed6cb24d0ef448b0c0b0239607cecbd225cb19f1b40ad604cc33e8
    File: /Users/nsowatsk/Library/Caches/Homebrew/downloads/dc411894f2c180133d2ff4aa9ed75117b145ffba55c0e7b176240f19427d4b2b--attachment_331173.xz
To retry an incomplete download, remove the file above.
```

NathanDotTo, 2021-08-01:
After downloading `https://bugzilla.gnome.org/attachment.cgi?id=331173` manually, I tried:

```
sha256sum ~/Downloads/attachment_331173.xz
b542466d48ed6cb24d0ef448b0c0b0239607cecbd225cb19f1b40ad604cc33e8  /Users/x/Downloads/attachment_331173.xz
```

I am note sure where the expected value is coming from, but something is not what it should be, clearly.

zhujuncode, 2024-03-02:
==> Fetching osrf/simulation/ogre1.9
==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff

xujunhai, 2024-12-31:
> ==> Fetching osrf/simulation/ogre1.9 ==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff

the same problem！！！